Ultimaker/Cura #16022: Display bug on infill when infill layer height significantly larger than overall layer height

kolergy, 2023-07-05:
### Cura Version

5.4.0

### Operating System

linux pop-os

### Printer

flsun v400

### Reproduction steps

layer height 0.05
infill layer height 0.3
all wall thickness 0.44

### Actual results

display extremely large infill wall thickness look like they have been multiplied by 6 the ratio between the layer height and the infill layer height.
print is OK though only a display bug

### Expected results

normal wall thickness in display

### Add your .zip and screenshots here ⬇️

[cube.zip](https://github.com/Ultimaker/Cura/files/11962496/cube.zip)

kolergy, 2023-07-06:
[image]
[image]

Asterchades, 2023-07-06:
I believe this is expected behaviour. Not necessarily intended, but not easily avoidable.

The preview system operates on individual layers, which are subsequently stacked on top of each other. The layers where the infill is "skipped" will subsequently show absolutely nothing as there's no GCode present to actually put anything there, where the layers which do have infill will now have a massive over-extrusion. Since the preview has no way of _easily_ determining that this over-extrusion is as a result of trying to make the infill taller, it will simply render it the same way it would if the Flow had been turned up by the same amount: making it wider.

Chances are any other GCode visualiser is going to do exactly the same thing for the same reason.

kolergy, 2023-07-07:
Surprising... Since you have definition of line width and height I would have assumed that is what would be used.

MariMakes, 2023-07-26:
Hey @kolergy,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

I agree with @Asterchades, this seems to be intended behavior for quite a while.
I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞

kolergy, 2023-07-26:
@MariMakes , 

Understood, it is just a display issue, not blocking anything.

MariMakes, 2023-08-02:
Hey @kolergy,

I showed your issue to the team. 
It seems to have been a conscious decision.
With the tools available it's the best way to show the wall thickness since it spreads over multiple layers but it's a single extrusion bead.  We also don't know a way to improve the visualisation for this.

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
I'll be closing this issue. I hope you understand.

closed 2023-08-02 by MariMakes

kolergy, 2023-08-03:
Well I understand that:
- It is just a display issue
- it can not be fixed easily, 
- it is not a big issue so it is clearly not a priority

But it is still a bug  though :p).