Ultimaker/Cura #16023: Missing some perimeters when using fuzzy skin

kolergy, 2023-07-05:
### Cura Version

5.4.0

### Operating System

linux pop-os

### Printer

flsun v400

### Reproduction steps

activate fuzzy skin

### Actual results

missing some perimeters in preview and on part

and some times some extra ones

### Expected results

the rights perimeters at the right place

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]

MariMakes, 2023-07-26:
Hey @kolergy,

Thanks for your report 👍 That looks odd!

I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

MariMakes, 2023-07-27:
Hey @kolergy,

Thanks for the file👍

It confirms my suspicion. To resolve some klipper issues in Cura 5.4 we itnroduced a smoothing algorythm that can be a bit too aggressive sometimes if you have a higher resolution. If you reduce the value it shows up as expected: 
[image]

We've added a ticket to the backlog with the intent to improve this and this project file is a great test file. Thank you for theat. 
For internal reference https://ultimaker.atlassian.net/browse/CURA-10856

Thanks for the report! 👍