Ultimaker/Cura #16027: CE3E3V2_make solid 3D model

katastropi, 2023-07-06:
### Cura Version

5.2.1

### Operating System

Windows

### Printer

Creality Ender-3/Ender-3v2

### Name abnormal settings

none

### Describe model location

it's located at the center of the plate

### Describe your model

Yes, I have sliced a model successfully before

### Add your .zip here ⬇️

[CE3E3V2_make solid 3D model.zip](https://github.com/Ultimaker/Cura/files/11963895/CE3E3V2_make.solid.3D.model.zip)

katastropi, 2023-07-06:
I've also downloaded the 5.3.1, however the same problem occurs

katastropi, 2023-07-06:
> Thanks for the report and the file.
> Go to the MarketPlace and load the Mesh Tools plugin. It has some repair capability. A more complete repair program is MS 3D Builder which you should find in the Microsoft App Store.
>
> When Cura puts polka-dots on a model it means the model has errors. The errors might be open seams (non-watertight) or in this case it looks like some surfaces are facing inside out. Both types of errors confuse slicers because the slicer can't tell the inside of the model from the outside. So this model needs repair.
>
> This is the repaired model. It sliced with no problem in 5.3.1. Opening your project file in 5.4 caused Cura to crash hard.
> I'm going to leave the bug label on this because of the crash. There doesn't seem to be a bug causing a slicing error, it's a bad model.

Alright. Thank you very much for the help

GregValiant, 2026-09-10:
I'll close this.

closed 2026-09-10 by GregValiant