Ultimaker/Cura #16030: new update problem

themr469, 2023-07-06:
### Cura Version

5.4.0

### Operating System

windows 10

### Printer

creality ender 3 s1 pro

### Reproduction steps

opend slicer to make some adjustments and test new version

### Actual results

printer "forgets" how it is supposed to work. prints mid air, doesnt do check from left to right front to back up and down. just starts printing where ever 

### Expected results

print like normal

### Add your .zip and screenshots here ⬇️

[CE3S1PRO_3clip2.1 v1.zip](https://github.com/Ultimaker/Cura/files/11968346/CE3S1PRO_3clip2.1.v1.zip)

MariMakes, 2023-08-08:
Hey @themr469,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I don't have an Creality Ender 3 S1 printer, but I think I know what might be wrong.
Your origin seems to be at the center. I don't know any Creality Printers that have set this setting like that. 
You can change it by going to the Machine Settings, and removing this checkmark. 
[image]

Can you let us know if removing the checkmark works for you? 
If it's not the case, we'll take another look.

themr469, 2023-08-08:
Hi. Thanks for your email
That worked, I don’t remember ticking that box. But now but now I see
something different that it actually works, I don’t know if it’s a problem
or what but the printer does not do the XYZ check before starting printing.
But it seems to be printing perfectly fine. Is that just a setting im
missing? Thanks again for your response.

themr469, 2023-08-08:
Update. It worked on the first print. But when i tried to print the same
thing again ( same file on sd card) it dit exactly the same as before.
Where it seems as if my printer got lost. I dont mind helping you do the
testing with my printer. Maby if you have a place where I can send you vids
of it  and exactly what it does?

MariMakes, 2023-08-08:
Hey @themr469,

If you reprint the same Cura Gcode and your printer behaves differently, it's not a Cura issue anymore. It seems to be something that has to do with your printer. Then it's better to consult people with the same printer as you do, like on the [Creality Reddit. ](https://www.reddit.com/r/Creality/)

Would you still like to keep this issue open or do you agree that we can close it?

themr469, 2023-08-08:
I only have this problem with the new update. The older version works 110%
everytime. I believe its a cura problem. As mentioned the first time a
print fine and every time there after it doesn’t

themr469, 2023-08-08:
After further testing. I have a new discovery.
When I do auto Level and print directly after, it works fine. Then if i
print again the printer gets “lost” doesnt do XYZ checks before starting,
it just starts where it left off.

When using older version of cura, i can print 100 one after the other, I
always get a XYZ check before it starts and it knows where it is.

Im not complaining about this, just trying to give my in put and help make
this better. Cura is the only slicer i have used and want to continue, but
I fear if I get such a rude response again….

Please do not close this case. Let me record a video of the problem and
sent it via google drive link.

MariMakes, 2023-08-09:
Hey @themr469,

I took another look, it might have to do with your start gcode for the creality ender 3 s1 pro.

Can you change the following thing in your start gcode:
Remove `"G28 O ;Home all untrusted axes" `and replace it with `"G28 ; Home all axes"`

This change was introduced by default a few weeks ago:
https://github.com/Ultimaker/Cura/commit/7f04c326fbb0144d8cfd76a2ae66932f09ad4e48 
This fix will be included in the next release of Cura but is not available in te version that you are running. 

Can you let us know if that solves your problem?

themr469, 2023-08-11:
> 7f04c32

Good day. So i did that and my printer did the XYZ Checks but then it went
to the front left and did nothing while showing print progress on screen

MariMakes, 2023-08-15:
I'm glad that at least you are no longer stuck. 
You are right, there is a significant difference between these two. 
But at first glance, there is nothing alarming. 

[image]

I'll show it to the team, perhaps they see something that could cause your printer to do the XYZ Checks but then it went
to the front left and do nothing while showing print progress on the screen🤞

MariMakes, 2023-08-25:
Hey @themr469,

I showed your issue to the team, but they couldn't see anything odd, other than perhaps on Line 2, where a command has been edited out.  Unfortunately I'm also stuck, and wouldn't know how to help anymore than this. 

@GregValiant has more experience with Creatlity printers and start gcodes. Perhaps he can see what is wrong.

GregValiant, 2023-08-29:
Good morning.
There was a pull request that changed the StartUp Gcode to  include this line:
"G28 O ;home all untrusted axis"
The "O" parameter is supposed to tell the printer to home any axis that has lost it's position, but not all firmware supports G28 with an "O" parameter and printers were hanging up on it.  But I don't see that problem in your StartUp so I'm with Mari on this, I don't see anything odd in there either.
One thing that is not present is the Auto-Level command (for Creality it's usually "G29" but can be "M420") so if the machine does indeed have an ABL it isn't being called.  It would typically be placed as the line after G28.