Ultimaker/Cura #16034: Preview of Infill Layer Thickness increase seems incorrect

rondlh, 2023-07-06:
### Cura Version

5.4.0

### Operating System

Windows 10 x64

### Printer

Custom

### Reproduction steps

Compare the result of 1x the layer height and 5x the layer height in the preview

### Actual results

Infill layers seems to get much wider, wall surfaces are visually affected too. I believe this is only an visual issue (didn't test it), I assume the actual print is not affected like shown.

Infill layer thickness 0.05mm ( = layer height), everything looks normal.
[image]

Infill layer thickness 0.4mm (8x the layer height), infill seems far too wide. Some artifacts in the walls are visible.
[image]

Note: The used material is about the same.

### Expected results

Visually there should not be much difference.

### Add your .zip and screenshots here ⬇️

[image]

Asterchades, 2023-07-07:
Duplicate of #16022.

Much as I explain there, I believe this behaviour is unintentional but largely unavoidable due to the nature of the preview system.

closed 2023-07-07 by Asterchades

ymh1003, 2023-12-05:
Can you share the zip file? Much appreciated!