Ultimaker/Cura #16035: "Select face to align to the build plate"feature doesn't work

vdd9474a, 2023-07-06:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 5 S1

### Reproduction steps

1- Drag and drop an stl file to add it in Cura
2- Select "rotate"and "select face to aligm to the build plate"
3- click on the face that I want to align to the build plate

### Actual results

The model rotate unexpectedly, the face that I selected isn't aligned to the build plate. The face that I want to align is flat.

### Expected results

The face I have selected flat against the buildplate

### Add your .zip and screenshots here ⬇️

https://github.com/Ultimaker/Cura/assets/10398512/2c5b225f-30ba-4ae5-a525-4568d12effb0

LeanderVanhulle, 2023-07-07:
I can confirm this bug since 5.3.1. Using Windows 3d builder to 'settle' the object on the correct side, save and import then in Cura. This as temporary patch until this issue is fixed.

MariMakes, 2023-07-26:
Hey @vdd9474a,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Unfortunately, the video that you shared is not visible to me. 
Are you trying to arrange a group of objects?
Because that's a duplicate of https://github.com/Ultimaker/Cura/issues/14237

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-08-09 by github-actions[bot]