Ultimaker/Cura #16038: Failed slice. I am new to FDM, having come from high-end sla printing. I'm having a lot of fun.

GregECGS, 2023-07-06:
### Cura Version

5.3.1

### Operating System

Windows 10

### Printer

Ultimaker 2+e

### Name abnormal settings

I turned on Adaptive Layers from 'Experimental'. Turning it off didn't help

### Describe model location

both parts sliced as they imported

### Describe your model

first print of this item. I designed it as a 2 part spool holder to hang off the side of my printer

### Add your .zip here ⬇️

[SPOOL_HOLDER_A1-SIDE-B.zip](https://github.com/Ultimaker/Cura/files/11973025/SPOOL_HOLDER_A1-SIDE-B.zip)

GregValiant, 2023-07-07:
Thanks for the report and the file.
I didn't have any trouble slicing it with or without adaptive layers so I'm unable to duplicate the problem.  I'm also on Win10.

There is a persistent bug that started in 5.3.0 that can keep good models from slicing.  There are a couple of workarounds for that.
- Try moving the models to different areas of the Cura build plate.
- Try rotating the models on the build plate.
- Sometimes changing the Initial Layer Height by just a bit (Ex: from .27 to .23) can fool the bug.

I'll leave the bug label on this for the Cura team.
[image]

GregECGS, 2023-07-07:
Thank you for taking the time to look at this. I was able to get them to print one at a time, so there isnt a problem on my end. I would love to know what caused the issue.
Have a great day.

closed 2023-07-07 by GregECGS

GregValiant, 2023-07-07:
Every mesh is made up of triangles.  The higher the resolution of the model file, or the higher the complexity of the model increases how many triangles a slicer has to section to get the shape.
I think this bug is about how Cura "sees" the file when it slices it.  Whether moving, rotating, or changing the initial layer height, they all move the position where Cura starts the slicing.  I think the bug just hits a spot that it can't figure out and gets stuck.  When slicing it from another direction or at a slightly different height, the problem is missed and the slice finishes.
That's my theory anyway.  The Cura team has been working on it, but it isn't fixed yet.

GregECGS, 2023-07-07:
Thank you for your time, Greg.

Attempting to resolve this, I managed to get enough of my settings hosed up that I had to factory reset my printer and the Cura software. Now that I know moving the parts around helps, I don’t think this will be a serious issue. At least, not for me.

Have a great weekend, Greg.

Regards,
Greg Mitchell

566 Airport Rd
Louisburg, NC 27549
919-672-2705
***@***.******@***.***>
[LOGOS]

EXTERNAL EMAIL

GregValiant, 2023-07-07:
People should not have to fool with the model in order to get it to slice.  It wasn't necessary until 5.0 but since then Cura has been temperamental about the way that certain models are orientated.