Ultimaker/Cura #16044: Slicing failed

JesConsuegra, 2023-07-07:
### Cura Version

5.4.0

### Operating System

MX Linux

### Printer

Hypercube Evolution

### Name abnormal settings

none, it worked fine so far

### Describe model location

centered in the platform

### Describe your model

a picture frame

### Add your .zip here ⬇️

[frame.zip](https://github.com/Ultimaker/Cura/files/11980465/frame.zip)

GregValiant, 2023-07-07:
Thanks for the report and the file.
Here is a report on your model from formware.co/onlineSTLrepair
-> Reading file and indexing vertices
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
**--> 26324 Non-manifold edges (?)
--> 90295 Inverted faces (?)
--> 1260 Degenerate faces (?)
--> 1731 Duplicate faces (?)**
--> 0 Disjoint shells (?)

Here is the view in Mesh Mixer.  The errors seem to relate to those fancy fleur-de-lis in the corners and their adjacent radii.
[image]

Here is the model after being repaired in MS 3D Builder (the formware site timed-out) as sliced in 5.4.0.
[image]

I'm going to remove the bug label and close this.  The model needs help and after repairing the model the behavior went away.  I hope you understand.

closed 2023-07-07 by GregValiant

JesConsuegra, 2023-07-07:
Thanks, anyway Cura 5.3 was able to process it....