Ultimaker/Cura #16052: Unable to load multiple files

justinroyboulanger, 2023-07-07:
### Cura Version

5.4.0

### Operating System

Windows 10 

### Printer

CR-10 pro

### Reproduction steps

Open file previously could open multiple different files at once and arrange them on the build plate. 

With the new 5.4 I need to open file for each different model i'll be placing on the build plate.

Steps :
 File > Open file(s) (ctrl+O)
 Try to select more than 1 .stl file 

### Actual results

Only one file can be selected. Manually entering the Path of multiple files like "file.stl" + "file.stl" still works.

### Expected results

Multiple files should be selected

### Add your .zip and screenshots here ⬇️

[image]
[image]

GregValiant, 2023-07-08:
Thanks for the report.
This bug has been reported previously ( #15879 and #16003 ) so I'll mark this as a duplicate and go ahead and close this.  I hope you understand.
You can still drag and drop multiple files but the "Open File" dialog doesn't allow for multiple files.

closed 2023-07-08 by GregValiant

MikeOxawopper, 2023-07-08:
having the exact same issue, reported in beta, still present in release version
cant control click more than one part
Cant box select more than 1 part
cant control a to select all

plus slicing is slower
and the same test print the pen holder in 5.4 comes in at 7 hour longer
[image]
[image]
[image]
[image]