Ultimaker/Cura #16053: ducks

ItsJohnKelly, 2023-07-07:
### Cura Version

5.4

### Operating System

win 11 pro

### Printer

ender 5 pro

### Describe model location

its close to the maximum size

### Describe your model

pen holder.

### Add your .zip here ⬇️

[duck-v6.zip](https://github.com/Ultimaker/Cura/files/11984051/duck-v6.zip)

GregValiant, 2023-07-07:
Thanks for the report.
I can see that the gcode file looks bad but without a project file there isn't any way to tell why.  My first guess is that it's a bad model.

Load the model.  Set Cura up to slice.  Use the "File | Save Project" command to create a 3mf file.  Zip the 3mf file and post the zip folder here.

ItsJohnKelly, 2023-07-07:
Hello,I’m afraid I have moved on and it now slices ok.  I have sliced and printed this a few times in the past with no problems, but never with the latest version of cura. The changes made since that report include rotating the model a little around the z-axis (so as to make the two ends point towards the respective corners more accurately) and I have also altered the orientation a little (25° along x-axis) and then made the edge of the base lie flat.  I also slowed the raft print speed to 30mm/sI do not seem to be having any problems with it at the moment…it’s printing fine.Best wishes,			John Kelly

GregValiant, 2023-07-07:
I'll go ahead and close this then.  Good luck.

closed 2023-07-07 by GregValiant

ItsJohnKelly, 2023-07-07:
Hello, Many thanks for your fast response. :)Best wishes,			John Kelly