Ultimaker/Cura #16054: Shouldn't this have generated an error or at least a notification?

fayena, 2023-07-07:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Modified Ender 3

### Reproduction steps

1. Load model 
2. Slice 
3. notice that the bottom of the print is very thin
4. check model in Cura    

### Actual results

Model is loaded below the build plate

### Expected results

Generate error or notification that part of the model is below the build plate.   

### Add your .zip and screenshots here ⬇️

https://www.printables.com/model/175450-hat-loom-knitting/files
[image]

GregValiant, 2023-07-07:
Thanks for the report.
There are definite situations where adjusting a model below the build plate is desired.  I've done it a lot both for special effect, or to turn a large model into separate pieces that I can glue together.

If you go to Preferences/Configure Cura - on the General tab you will see "Automatically drop models to build plate."  When that is enabled then all models should load into Cura with the lowest most point at Z=0.

I'll leave the bug label on this and you can respond, but this appears to be expected behavior.

fayena, 2023-07-07:
Thank you for the response.   I just checked, and that option is already selected.     

Maybe a notification to let the user know?   It was just a silly mistake, but I've never had that happen before.   The file loads at the build plate in SuperSlicer.

GregValiant, 2023-07-07:
Was that an STL file or some other model format?  Some 3mf model files have location information contained in them.  Cura will respect that location information.  I've had models come in at all sorts of Z heights and well off the build surface.

fayena, 2023-07-07:
1. It was a .3mf     https://www.printables.com/model/175450-hat-loom-knitting/files   

But, like I said, there was no warning that some of the print was below the build plate, and superslicer sliced it at the build plate.

GregValiant, 2023-07-08:
That model does have location data in it.  Whether I have "Automatically drop..." on or off - the model comes in at -9.08mm.
The Cura team will take a look.

MariMakes, 2023-07-26:
Hey @fayena,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

@GregValiant is correct. The biggest difference between a .3mf and a .stl is that the 3mf will remember the location in a 3D space when you load it. In this case, it's saved below the buildplate. 

If a model is loaded below the buildplate we try to give it a distinctly different color like this
[image]

We have a lot of users, like myself, who do this on purpose. Giving a warning/notification would just hinder them.

To me, it doesn't seem like a bug in Ultimaker Cura anymore, but a problem with the file. 
Would you still like to keep this issue open or do you agree that we can close it?