Ultimaker/Cura #16056: Unable to connect to moonraker with 5.4.0

Myklt8, 2023-07-07:
### Cura Version

5.4.0

### Operating System

Windows11

### Printer

Creality Ender 3

### Reproduction steps

Sliced the file
upload to printer
takes a long time but then times out

### Actual results

Fails to connect to moonraker
[image]

### Expected results

should have uploaded to the printer
I am seeing this issue a lot on FB and groups are telling people to roll back

### Add your .zip and screenshots here ⬇️

No Data log exists

MariMakes, 2023-07-26:
Hey @Myklt8,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

As an UltiMaker employee, I don't have access to Moonraker 3D Printers, so I can't really help with troubleshooting your issue. Could it be related to the plug-in you are using? Are you using the latest version? You can download it here: https://github.com/emtrax-ltd/Cura2MoonrakerPlugin

To me, it doesn't seem like a bug in Ultimaker Cura.
Apologies for any inconvenience caused by our delayed response. 
I'll be closing this issue. I hope you understand.

closed 2023-07-26 by MariMakes

Casperdme, 2023-08-25:
I have the exact same issue.

MariMakes, 2023-08-28:
@Casperdme, don't forget to reach out to the plug-in creator here https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/issues/new

If it turns out to be an issue in Cura, I'm happy to take another look

Casperdme, 2023-08-28:
Thank you for this information.