Ultimaker/Cura #16057: Print order within layer. Start point of next layer.

Diiescher, 2023-07-07:
### Cura Version

5.4.0

### Operating System

Window 10

### Printer

Tronxy xy2

### Reproduction steps

1) Load Model: 
Articulated Christmas Dragon by P1lotz on Thingiverse: https://www.thingiverse.com/thing:5676934
(GIF example  is the "extra small" variety)
2) Slice it.

### Actual results

Print order in layer is weird.
Very good to see in Layer 10 (Fine Quality / 0.15mm layer hight), where "inner joint" is often printed before "outer joint"

This leads to a lot of unnecessary travel and potential stringing + prolonged print time
[image]

### Expected results

I would have expected the layers to be printed more in one go, rather than in this rather irregular order.
Also, the starting point for the new layer often does not seem to be wisely chosen, resulting in another unnecessary travel.

### Add your .zip and screenshots here ⬇️

cannot upload project file: "File size too big: 25 MB are allowed, 67 MB were attempted to upload"

GregValiant, 2023-07-07:
Thanks for the report.
I downloaded the file and sliced the "Small" version with my normal settings for my Ender 3 Pro.  I don't show that kind of travel on the first few layers up to layer 7.  The print starts at the head and goes back to the tail.  On layer 8 (with my settings) the loops between the segments become islands and then the toolpath changes and there is more travel.
I had Combing set to "Not in Skin" and the Z seam set to "Right".   What did you have those two settings at?

The print time with my settings is 8:38.  7% of that is travel while 29% is retractions (which you can't do much about).  You might be able to cut the travel in half by changing the toolpath, but on another model it would likely be disadvantageous.

I don't see a bug here but I'll leave the label on and someone from the Cura team will take a look.

Diiescher, 2023-07-08:
Combing is also set to "Not in Skin" 
For Z seam alignment I dont see a "right" setting. (it is set to "sharpest corner")

For me the strange sorting starts at layer 4, where print order is: 
1) From middle to tail 
2) middle to head, skipping some segments-
3) +4) going reverse to missed segments.

[image]

Print time is far more, but this might be because of me beeing rather new to 3D printing and still looking for best setting.

[image]

GregValiant, 2023-07-08:
If you set the "Z seam alignment" to "User Specified" the other options are available and you could also enter your own XY numbers.
The "bug" label is still in place on this and the Cura team will take a look.
Those "Time Estimates" are based on your current settings which I'm sure are different than the ones I used.  That's why you get different estimates.  Looking at your numbers and at the model, an hour spent traveling really doesn't seem like and inordinate amount of time.  The possibility of stringing is a concern but looking at all the islands that develop as the print advances there are going to be a lot of retractions.  You want to configure the "Max comb distance with no retract" to a small number to insure that whenever the nozzle leaves and island there is a retraction or yes, there will be a lot of stringing.  If a travel move is shown as a heavy blue line then there was no retraction for that move.

Load the model and set Cura up to slice.  Use the "File | Save Project" command.  In File Explorer right click on the 3mf file and select "Send To / Compressed (Zipped) folder" and post the Zip folder here.

Diiescher, 2023-07-08:
"user specified"!... I learned something. Thanks :)

Project is saved as it was for the screenshots. File took around 30sec(!) to save (on an AMD 5800X/32GB to M2 SSD) and is still too big for upload. 
So I had to trick the forum software. :) 
This is the ".3mf" zipped into 3 parts. You need to delete the ending ".zip" and exctract with something like "7zip" that can handle multi part archives.

[TXY2PROPRO2EXY3SE_xmas_dragon_extra_small_-_P1lotz.zip.001.zip](https://github.com/Ultimaker/Cura/files/11990620/TXY2PROPRO2EXY3SE_xmas_dragon_extra_small_-_P1lotz.zip.001.zip)
[TXY2PROPRO2EXY3SE_xmas_dragon_extra_small_-_P1lotz.zip.002.zip](https://github.com/Ultimaker/Cura/files/11990621/TXY2PROPRO2EXY3SE_xmas_dragon_extra_small_-_P1lotz.zip.002.zip)
[TXY2PROPRO2EXY3SE_xmas_dragon_extra_small_-_P1lotz.zip.003.zip](https://github.com/Ultimaker/Cura/files/11990622/TXY2PROPRO2EXY3SE_xmas_dragon_extra_small_-_P1lotz.zip.003.zip)

No dark blue lines, all "hops" are properly retracted. 
Playing around with "Z Seam Alignment: User Specified" seems to slightly improve print order, but to me it still looks "bugged".