Ultimaker/Cura #16058: Cura causes Bluetooth + System crash (macOS)

Careless, 2023-07-07:
### Cura Version

5.4.0 & 5.3.0-beta1

### Operating System

MacOS Big Sur 11.7.8

### Printer

Custom Printer Profile

### Reproduction steps

1. Have Bluetooth Dongle in USB Port + Headphones Connected
2. Open Cure
3. Wait for a minute approx 60 sec.
4. Bluetooth Error shows up
[image]
5. Close Error Dialog Box via [  OK  ] Button
6. Immediately Upon quitting CURA, computer panics and reboots.

### Actual results

Computer will reboot into macOS and display "error caused computer to shut down unexpectedly" dialog box, typical of macOS hard crashes.

### Expected results

Solution:

As expressed in this reddit thread here, which I replied to 3 years ago:
https://www.reddit.com/r/3Dprinting/comments/gt61et/cura_461_causes_bluetooth_and_sound_issues/

This seems to be an issue since 4.6.1

The solution, as pointed out in the first post, does seem to fix the problem. I currently have CURA open right now with the USB Printing Plugin disabled.

If I re-enable this plugin, and perform the steps again, the computer will crash, 100% of the time.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-07-07:
Thanks for the report.
The only thing I can think of that could do this is the Auto-Polling of the USB Printing Plugin.  It looks for USB ports configured as serial ports (note your warning dialog)  on any USB port and then does a check using all popular baud rates.  It sends M105 to check the temperature and if there is no response it moves on to the next baud rate.  This is an ongoing process when USB Printing is enabled in the MarketPlace.

Have you disabled USB Printing in the MarketPlace?  It is "enabled" by default.

Careless, 2023-07-07:
Hi,

Thanks for your reply. 

Yes, as I have stated in the 'Solution' input box on the issue form, the way to fix it (I guess), is to disable the USB Printing Plugin in the marketplace.

How can I ensure this isn't re-enabled on future updates?

Careless, 2023-07-07:
If this is automatically being turned on during updates, then is there a way for CURA to look at the current settings and disable the plugin during updates?

Or is it supposed to already do that, and there is a bug somewhere?

I should mention that I recently upgraded to 5.3.1 and then closed the program because I wasn't printing anything. But I have never experienced his issue since 4.x.x... And I don't remember disabling the plugin... So something changed.

GregValiant, 2023-07-07:
Not for that particular plugin.  Whenever we update Cura it has to be shut off.  I print via the SD card but I have a piece of software to control it via the USB.  Cura interferes because it grabs the port if USB Printing is enabled.  I always have to disable on a new install.  It's two mouse clicks so it isn't an onerous task.

Careless, 2023-07-08:
> Whenever we update Cura it has to be shut off.

>I always have to disable on a new install. It's two mouse clicks so it isn't an onerous task.

Except it causes a kernel panic, which can cause the user to lose data? Seems like a problem, if that's the case.

Why couldn't there be a checkbox option in the preferences to scan for devices upon opening cura, so that even if it's turned on it's not hijacking USB ports.

I've literally never experienced an issue like this in using any piece of software that connects to a USB device on macOS, so whatever CURA is programmed ot do here is not normal.

fieldOfView, 2023-07-08:
Disabled plugins are registered in the Cura configuration folder (under Library/Application Support/cura/version on mac OS). A new version of Cura will make a copy of the most recent version of the configuration folder and work with that copy. So the disabled state of plugins should carry over to new versions of Cura.

Careless, 2023-07-08:
> 

that's the action I expected and presumed, but at some point it must have turned on again? 

Not sure how, I've had the same configs since 2011 for the most part, Including a lot of your plugins, Aldo.

Is it possible the startup optimizer plugin changed something?

[image]

Not accusing anyone of anything, just trying to understand what it's a problem all of a sudden...

And even if it did turn on at some point, it should not cause system crash. 

Claiming that "the crash won't happen if you just keep the plugin off" is not really a good way to address a kernel panic creating bug.

GregValiant, 2023-07-08:
I've never used the StartUp Optimizer (I load a lot of project files).  I have had to turn USB Printing off manually whenever I've installed a new version of Cura.  That was true for 5.4.0 install as well.

I have a USB multi-drive device and a USB mouse.  It would appear that they are not considered to be "serial devices" as Cura has never interfered with either of those and has consistently bypassed both while going ahead and commandeering the "USB to Serial" port that the printer is is plugged into (provided I forgot to turn USB Printing off which is the usual case).

Careless, 2023-07-08:
I am not using a USB connection to the printer though, so I can't speak to whether CURA detecting a printer early in this process changes its behaviour, but since 4.6.1 as illustrated in the reddit post I linked, this seems to have some effect on other USB devices.

can't say i've ever had another application that searches for USB devices take out my bt4.0 dongle and then warn me that my computer is about to crash.

this is a bug.

Careless, 2023-07-26:
thanks for the follow up

🤞