Ultimaker/Cura #16061: Slicing failed when changing .3mm layer height from .2mm layer height

sarbanharble, 2023-07-07:
### Cura Version

5.4.0

### Operating System

MacOS Ventura

### Printer

Ultimaker S5

### Reproduction steps

1. Successfully sliced model with layer height at .2mm on Extruder 1 with a .6mm CC Core.
2. Changed layer height to .3mm from .2mm layer height
3. Clicked "Slice"

### Actual results

Received "Slicing Failed" error

### Expected results

I expected to get slicing results. If I change layer height back to .2mm, then slicing completes successfully. If I change to .25mm layer height, it completes successfully.

### Add your .zip and screenshots here ⬇️

[image]
[CE3_fox mask.3mf.zip](https://github.com/Ultimaker/Cura/files/11986710/CE3_fox.mask.3mf.zip)

MariMakes, 2023-07-26:
Hey  @sarbanharble,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 
That's a pretty cool model 🤩

I think I figured out a way to get you unstuck.
Your model seems to be floating above the buildplate, the bottom isn't really flat. But I was able to slice your model with the 0.3 Layerheight when I dropped it a little in the buildplate. 
[image]
I hope this helps 😇

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

sarbanharble, 2023-07-26:
Well that makes sense! That would be the issue, as the model before I had turned off “auto place on the build plate”

Thank you much!

Barnabas Helmy
217-761-9292
barnabashelmy.com

MariMakes, 2023-07-27:
Thanks for the update. 
I'll close the issue now. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-07-27 by MariMakes