Ultimaker/Cura #16062: Failed to execute script 'cura_app' due to unhandled exception

aayevtushenko, 2023-07-08:
### Cura Version

5.4

### Operating System

Windows 10

### Printer

NA

### Reproduction steps

1. Download the app
2. Run the app

### Actual results

Failure to run the application with error:

Failed to execute script 'cura_app' due to unhandled exception: DLL load failed while importing QtWidgets: The specified procedure could not be found.
```
Traceback (most recent call last):
  File "cura_app.py", line 28, in <module>
  File "PyInstaller\loader\pyimod02_importers.py", line 352, in exec_module
  File "cura\CrashHandler.py", line 24, in <module>
    from PyQt6.QtWidgets import QDialog, QDialogButtonBox, QVBoxLayout, QLabel, QTextEdit, QGroupBox, QCheckBox, QPushButton
ImportError: DLL load failed while importing QtWidgets: The specified procedure could not be found.
```

### Expected results

Run the application

### Add your .zip and screenshots here ⬇️

NA

aayevtushenko, 2023-07-08:
updating the OS to the latest version resolved the issue.

closed 2023-07-08 by aayevtushenko

webguy24, 2023-08-24:
hi i have this isssue

RadianTOR, 2023-09-10:
Windows 8.1
I have this issue!

webguy24, 2023-09-10:
I had 8.1 also

RadianTOR, 2023-09-10:
> I had 8.1 also

I found information about Cura no longer support Win 8.1 :-(

webguy24, 2023-09-10:
time to switch to prusaslicer then lol