Ultimaker/Cura #16063: Iron Man Helmet Slicing Failure/SnapMaker A350

TKDMarine, 2023-07-08:
### Cura Version

5.4.0

### Operating System

Windows 10 Home, Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz, 32GB Ram

### Printer

Snapmaker A350

### Name abnormal settings

Not that I am aware of

### Describe model location

Model slices if moved from center of build plate

### Describe your model

Have not sliced model before. It is an Iron Man Helmet

### Add your .zip here ⬇️

[S2A_Iron Man MK3 Helmet.zip](https://github.com/Ultimaker/Cura/files/11990082/S2A_Iron.Man.MK3.Helmet.zip)

GregValiant, 2023-07-10:
Thanks for the report.
This sliced for my with no changes to your project.
You can try rotating it, changing the Initial Layer Height by a little bit, or moving it.
The Cura team will take a look.  The model does have some errors and is not watertight.
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic