Ultimaker/Cura #16064: Slicing Failed

inoble1, 2023-07-08:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 3 S1 Pro 

### Name abnormal settings

Won't slice when I increase the scale.

### Describe model location

No

### Describe your model

I have printed the model twice already

### Add your .zip here ⬇️

[CFFFP_3.zip](https://github.com/Ultimaker/Cura/files/11990254/CFFFP_3.zip)

GregValiant, 2023-07-10:
Thanks for the report.
There are some strange things going on in that model file.
[image]
Notice the surfaces below the build plate...There are several of those inside the model as well.  There are a total of 35 pieces on the build plate.
You can see the Cura warning that the "Model is Not Watertight and may not print correctly".
I see you have printed this before but that model needs repair.
You can try moving it on the build plate.  Since scale is a problem you can try changing the Initial Layer Height by a little bit

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic