Ultimaker/Cura #16070: Thin Walled tube fails in Cura

ProtomakerSprint, 2023-07-09:
### Cura Version

5.3.1

### Operating System

Windows 11

### Printer

Creality Ender 5 Plus

### Name abnormal settings

Error

### Describe model location

Vertical

### Describe your model

Thin Walled tube

### Add your .zip here ⬇️

How
[Gap Filler.zip](https://github.com/Ultimaker/Cura/files/11994054/Gap.Filler.zip)

GregValiant, 2023-07-10:
Thanks for the report.
The wall thickness is only .18 to .21 mm so you would need a tiny nozzle to print this.  Even with a .20 nozzle I don't get complete slicing.
You didn't really explain the problem you were having but no, this will not slice with a 0.40 nozzle.

This is with a .15 nozzle at .15 line width.  No problem except that my Ender 3 Pro can't print it.
[image]

I'll remove the bug label from this.  Let me know if you feel that is premature.

GregValiant, 2026-09-10:
I'll close this.

closed 2026-09-10 by GregValiant