Ultimaker/Cura #16072: Creality Sonic Pad Ender 3 S1 Pro    Profile wont import

LaurenceCarroll, 2023-07-09:
### Cura Version

5.4.0

### Operating System

Windows 10 64bit

### Printer

Creality Ender 3 S1 Pro

### Reproduction steps

[image]

[image]

Hope you can see the screen grabs.....  it says the profile failed to load due to "quality type "fast" is not compatible with the current active machine definition 'creality_base' 

This is one of 4 Profiles Creality supply with their Sonic Pad for this printer

### Actual results

As above after ctrl J to import the profile 

### Expected results

Profile should import?

### Add your .zip and screenshots here ⬇️

na

GregValiant, 2023-07-09:
Thanks for the report.
Creality has their own fork of Cura.  Have you tried importing the profiles into it?
This is for Ultimaker Cura and your printer definition and quality definitions may well be different here than than they are in the Creality slicer.
I'm going to remove the bug label for now.  Let me know if you feel that is premature.
Those profiles may work with the Creality slicer.  You can try creating a project file in Creality and importing that into UM Cura.  You may still get an error though.

KimiBobtail, 2023-07-09:
I followed the instructions to create a Custom FFF printer by manually writing the start and end Gcode and then importing the Cura profiles and it seems to work. (as mentioned in the "Creality Sonic Pad Ultimaker Cura（5.1.0）Profiles User Guide" pdf file)
(Setting>Printer>Add Printer>Printer Non Ultimaker>Add Printer Offline>Custom>Custom FFF printer)

LaurenceCarroll, 2023-07-10:
Hi Greg,

My apologies as I found out later that I needed to add the printer as a
Custom FFF one with definitions shown on a pdf file in the official
Creality Sonic Pad group on Facebook.

Once I did that the Profiles loaded in fine.  I couldnt figure out how to
get back to this too let you know.

Its great now I can use Cura 5.4.0 now.

Regards.   Laurie

GregValiant, 2023-07-10:
It's good you got it to work.  I'll go ahead and close this then.

closed 2023-07-10 by GregValiant