Ultimaker/Cura #16073: Slicing not working

scrody, 2023-07-09:
### Cura Version

5.4.0

### Operating System

windows 11

### Printer

Elegoo Neptune 3

### Name abnormal settings

Profile extra fast on default

### Describe model location

no

### Describe your model

yes it has sliced before no its not watertight no i haven't tried a mesh fix

### Add your .zip here ⬇️

[EN3_Raptor-Claw 2.zip](https://github.com/Ultimaker/Cura/files/11994421/EN3_Raptor-Claw.2.zip)

GregValiant, 2023-07-09:
Thanks for the report.
I didn't have any trouble slicing this "as recevied" in 5.4.0 so I'm unable to duplicate the problem.
You could try a couple of things
Change the initial layer height but just a bit.
Move the model on the build plate.
Rotate the model.
One or a combination of those may fool the bug.
The Cura team will take a look.
I changed the supports for this slice.  Your's would fail.
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic