Ultimaker/Cura #16074: raft does not cover the full object area

goofyseeker311, 2023-07-09:
### Cura Version

5.4.0

### Operating System

Win11 RTX3060

### Printer

elegoo neptune 3 pro

### Reproduction steps

trying to slice the project file with raft only (not supports).

### Actual results

does not raft the whole object area, leaves parts of the model without raft.

### Expected results

should raft selectively under the whole object area with default settings

### Add your .zip and screenshots here ⬇️

[image]
[EN3PRO_fsfsd3-v13-chassis.zip](https://github.com/Ultimaker/Cura/files/11994951/EN3PRO_fsfsd3-v13-chassis.zip)

goofyseeker311, 2023-07-09:
it will slice correctly if using raft + supports touching bed, which should be the default raft + no supports behaviour.

[image]

fieldOfView, 2023-07-09:
I am 99% sure the bottom of the model is not 100% flat. Looks like the circular sections start just a little higher than the rectangular section, even if the difference is less than the first layer height.. By adding supports, the circular sections are extended downwards and touch the buildplate all the way.

goofyseeker311, 2023-07-09:
its cad flat

MariMakes, 2023-07-26:
Hey @goofyseeker311,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I'm agreeing with @fieldOfView, the model itself is not flat. 
To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it? 

If  you run into another issue like the numbering of your profiles, so we can track it separately in our issue tracker.

closed 2023-08-09 by github-actions[bot]