Ultimaker/Cura #16077: Dual Nozzle Ender 3 set up Slicing Failed

Andrewthe0, 2023-07-09:
### Cura Version

5.4

### Operating System

Windows 10

### Printer

Custom Ender 3

### Name abnormal settings

Dual Nozzle Set up

### Describe model location

Towards Front Left

### Describe your model

Dual Object Merged Togethered

### Add your .zip here ⬇️

[CFFFP_Blue Divot Tool.zip](https://github.com/Ultimaker/Cura/files/11995308/CFFFP_Blue.Divot.Tool.zip)

GregValiant, 2023-07-09:
Thanks for the report.
This would absolutely not slice for me in 5.4.0 and I don't find any problems with the model.
It sliced fine in 4.13.1.  The Cura team will take a look.

Andrewthe0, 2023-07-22:
> Thanks for the report. This would absolutely not slice for me in 5.4.0 and I don't find any problems with the model. It sliced fine in 4.13.1. The Cura team will take a look.

Hey i just so happen to be looking through my settings just now and saw that the horizontal expansion setting was set to -0.1 so i changed that to 0 and now i am able to slice my model...... now i still dont know why i couldn't slice it because i could slice a dual color benchy from someone no problem.

GregValiant, 2023-07-23:
I just re-downloaded it and it sliced first try.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0