Ultimaker/Cura #16078: Skirt z-height

Diiescher, 2023-07-09:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

tronxy xy2

### Reproduction steps

Slice any model with skirt

### Actual results

strange skirt behaviour: 
1)  inner skirt line with 5.4.0 is three lines high for all "Skirt Line Count" >0
(5.3.1 does not have this) 
[image]

2) Setting "Skirt Line Count: 0" gives the same result as "Skirt Line Count: 2" + "Skirt Distance: 0"
(same in 5.3.1 but looks like "Skirt Line Count: 3")

[image]

### Expected results

Skirt does not have a height setting, so I would expect the skirt to have only one layer.
Setting "Skirt Line Count" to zero should have the same result as turning off "Build Plate Adhesion" 

### Add your .zip and screenshots here ⬇️

Thing: Tronxy XY 2 Pro Fan Vent by ChaacXib on Thingiverse: https://www.thingiverse.com/thing:4232479
[TXY2PROPRO2EXY3SE_excavator_extended.zip](https://github.com/Ultimaker/Cura/files/11995494/TXY2PROPRO2EXY3SE_excavator_extended.zip)

GregValiant, 2023-07-09:
Thanks for the report.
The feature is new in 5.4 and the setting is "Skirt Height".
The button to the right of the "Search Settings" box is the setting visibility control.  Set the visibility to "All".  Set "Skirt Height" to "1" to go back to the single layer skirt.
I'll leave this open for a bit in case you can't find the setting, but this is a new feature.

Diiescher, 2023-07-09:
should have read the changelog i suppose :)
thanks and sorry

GregValiant, 2023-07-09:
Better to report.  Closing these is as easy as hitting a button.
Have a good one.

closed 2023-07-09 by GregValiant

a4jp-com, 2025-04-16:
In Cura 5.10.0 are the first skirt lines always only the first layer height, except for the last line inside in Build Plate Adhesion which can be any layer height? Is there a way to have all the skirt lines 2 layers thick the the last line the height we want?