Ultimaker/Cura #16079: Unable to open files

anno0, 2023-07-09:
### Cura Version

5.4.0

### Operating System

Ubuntu 22.04

### Printer

Anycubic Mega X

### Reproduction steps

1. I want to open a file 

### Actual results

Cura 5.4.0 crashes

**Console output:**
./UltiMaker-Cura-5.4.0-linux.AppImage 
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltiMaQ9rFw3/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMaQ9rFw3/share/cura/resources/qml/Actions.qml"
file:///tmp/.mount_UltiMaQ9rFw3/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property "height"
file:///tmp/.mount_UltiMaQ9rFw3/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property "height"

(UltiMaker-Cura:12768): Gtk-WARNING **: 17:55:05.718: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Yaru/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Yaru/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
/tmp/.mount_UltiMaQ9rFw3/AppRun: line 25: 12768 Aborted                 (core dumped) $scriptdir/UltiMaker-Cura "$@"

### Expected results

the file dialog should be opened

### Add your .zip and screenshots here ⬇️

[image]

majekw, 2023-07-19:
Same here.
System: Slackware 15.0
It crashes on clicking 'open' icon or choosing File/Open File(s) from menu.
It also crashes on choosing File/Export option.
I could load files from recent file list.

Logs:
On start:

```
$ ./UltiMaker-Cura-5.4.0-linux.AppImage 
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltiMaYyC8fC/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMaYyC8fC/share/cura/resources/qml/Actions.qml"
qt.qml.context: file:///tmp/.mount_UltiMaYyC8fC/share/cura/resources/qml/Settings/SettingView.qml:427:16 Parameter "index" is not declared. Injection of parameters into signal handlers is deprecated. Use JavaScript functions with formal parameters instead.
file:///tmp/.mount_UltiMaYyC8fC/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property "height"
file:///tmp/.mount_UltiMaYyC8fC/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property
"height"
``` 
After clicking 'open icon':
```
(UltiMaker-Cura:9565): Gtk-WARNING **: 23:35:03.990: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Tango/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Tango/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
/tmp/.mount_UltiMaYyC8fC/AppRun: line 25:  9565 Aborted                 $scriptdir/UltiMaker-Cura "$@"
```

majekw, 2023-07-20:
I checked UltiMaker-Cura-5.4.0-beta.1-linux.AppImage and it also crashes the same way.
Older UltiMaker-Cura-5.3.1-linux.AppImage works fine.

gircysm, 2023-07-21:
(See also [15689](https://github.com/Ultimaker/Cura/issues/15918), [15918](https://github.com/Ultimaker/Cura/issues/15689))

Running into the same issue with `UltiMaker-Cura-5.4.0-linux.AppImage`.
Appears to be working fine with `UltiMaker-Cura-5.3.1-linux.AppImage` as noted above, but also fortunately seems fine with `UltiMaker-Cura-5.4.0-linux-modern.AppImage`.

MariMakes, 2023-07-26:
Hey @anno0,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 
We updated the PyQT version in Cura 5.4, I think what you are seeing is related to that. 

So I'm a Windows user but I've been told that you can try to `apt install xdg-desktop-portal-gtk`

You can set the `QT_QPA_PLATFORMTHEME `to one of the following values: 
- gtk2
- gnome
- kde
- qt5ct
- xcb
- xdgdesktopportal (current default).

Or start the Cura AppImage with the argument `-platformtheme ` with one of the following:
- gtk2
- gnome
- kde
- qt5ct
- xcb
- xdgdesktopportal (current default).

There could be more, possible values are scattered over the internet; Feel free to complete this list.

Can you let us know if that works for you? 

If that's not the case please let us know and we'll be happy to take another look.

majekw, 2023-07-26:
I installed xdg-desktop-portal-gtk (rebooted, made sure that process is running), but it didn't help.
Same for -platformtheme options, error message is always the same regardless of provided option.

strangeattraction, 2023-07-31:
UPD: [UltiMaker-Cura-5.4.0-linux-modern.AppImage](https://github.com/Ultimaker/Cura/releases/download/5.4.0/UltiMaker-Cura-5.4.0-linux-modern.AppImage ) works fine for me

anno0, 2023-07-31:
> I installed xdg-desktop-portal-gtk (rebooted, made sure that process is running), but it didn't help. Same for -platformtheme options, error message is always the same regardless of provided option.

That's the same behaviour for me. 

- The obviously reason is that /usr/share/icons/Tango/16x16/status/image-missing.png can not be loaded. So I installed tango-icon-theme. This has the effect, that I get the following error

`
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Yaru/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Yaru/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
`

Now `/usr/share/icons/Yaru/16x16/status/image-missing.png ` is reported that it can not be loaded.
This can not be true, because this image exists on my system and it's an iamge:

[image]

Wivik, 2023-08-27:
Hi,

Same issue with 5.4.0 and 5.3.1 for me :'( 

```bash
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Gtk-Message: 11:08:53.970: Failed to load module "xapp-gtk3-module"
Gtk-Message: 11:08:53.972: Failed to load module "canberra-gtk-module"
QObject::startTimer: Timers cannot be started from another thread
Cyclic dependency detected between "file:///tmp/.mount_UltiMazOYQ4Y/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMazOYQ4Y/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
file:///tmp/.mount_UltiMazOYQ4Y/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property "height"
file:///tmp/.mount_UltiMazOYQ4Y/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property "height"

(UltiMaker-Cura:4314): Gtk-WARNING **: 11:09:00.938: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /org/gtk/libgtk/icons/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /org/gtk/libgtk/icons/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
/tmp/.mount_UltiMazOYQ4Y/AppRun: line 25:  4314 Aborted                 (core dumped) $scriptdir/UltiMaker-Cura "$@"

```

OS is Manjaro. I usually install the AUR provided package because I'm not a fan of AppImages. But the error is present whichever I use the AppImage or the AUR package.

BTW : drag and drop a `stl` file still opens it and the tool remains usable so far.

Zedrackis, 2023-09-10:
Had this problem on LUbuntu/Xcfe 4.0. The solution was to use the 'modern' variant of Cura instead of the main branch. Works fine with that version.

sergeymarkow, 2023-09-11:
the same issue.

sergeymarkow, 2023-09-11:
> BTW : drag and drop a `stl` file still opens it and the tool remains usable so far.

but we still need to open save dlg to save gcode. The dlg produce the same issue...

sergeymarkow, 2023-09-12:
I found work around by changing theme for icons:
[image]

majekw, 2023-10-19:
Looks like this is fixed in 5.5.0-beta.1.
At least it works for me on Slackware 15.0

MariMakes, 2023-10-19:
That's good to hear 😊 . You can download the Cura version with the fix here: https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
I'll be closing this issue. I hope you understand.

closed 2023-10-19 by MariMakes