Ultimaker/Cura #16082: Slicing failed

hapklaar, 2023-07-09:
### Cura Version

5.4

### Operating System

WIndows 11

### Printer

Ratrig V-Core 3

### Describe your model

Yes on previous versions

### Add your .zip here ⬇️

[CFFFP_[shelled]An-Over-Engineered-Vase.zip](https://github.com/Ultimaker/Cura/files/11995818/CFFFP_.shelled.An-Over-Engineered-Vase.zip)

GregValiant, 2023-07-09:
Thanks for the report and the file.
I can't get this to slice in a 5.x version either.  There are a lot of errors in the model.  I repaired it using MS 3D Builder.

PrusaSlicer didn't have any trouble with it in Spiralize mode.  This is the repaired model.
[image]

Even Cura 4.13.1 didn't fare to well.  You can see all the gaps and this is also the repaired model.
[image]

The Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic