Ultimaker/Cura #16084: Cura Not Accepting Imports

JaxGM, 2023-07-09:
### Cura Version

05.04

### Operating System

MacOS Ventura

### Printer

Ender 3 v2 Neo w/ Octoprint

### Reproduction steps

I tried importing a file a few days ago and nothing happened. After drafting an issue report, the issue stopped so I never submitted it but the issue has returned.

### Actual results

I tried importing a file a few days ago and nothing happened. After drafting an issue report, the issue stopped so I never submitted it but the issue has returned.

### Expected results

The file should import properly.

### Add your .zip and screenshots here ⬇️

https://github.com/Ultimaker/Cura/assets/131479022/56deb0a9-0193-474f-b912-a10f66a01789

https://github.com/Ultimaker/Cura/assets/131479022/3218333b-3ba9-43ca-aff8-0092118bbd64

JaxGM, 2023-07-09:
It seamed to have resolved itself bit I believe it is the being of a larger issue.

JaxGM, 2023-07-10:
Update: The issue is happening again. After a few minutes of waiting with the app working, it finally imports.

MariMakes, 2023-07-26:
Hey @JaxGM,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Could it be that there is something wrong with your STL reader?
If you go to the Marketplace, Click the Gear and scroll down to see the STL Reader. If it looks like this you should click enable.
[image]

If that does not resolve your issue, please let us know. 
I'll take another look at what is going wrong. 👍

JaxGM, 2024-05-23:
Thanks for the help! It appears there is no STL Reader extension installed and reinstalling Cura has done nothing to restore that but the issue also occurs with other file types, such as 3mf, which has a reader installed. After about 10 minutes of the app being open, it begins to accept imports.

JaxGM, 2024-05-23:
Any suggestions are greatly appreciated!

JaxGM, 2024-05-28:
Update: I tried reinstalling again today but I deleted the saved settings for the first time. Now when I reinstalled it asked me to set up my printer but it still has a "cooldown" of sorts for imports :/