Ultimaker/Cura #16085: BUG

HEYheyheyrey, 2023-07-09:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

1. Slice failed

### Actual results

report bug

### Expected results

I could save it to the desk 

### Add your .zip and screenshots here ⬇️

[image]

HEYheyheyrey, 2023-07-09:
i need help fast, becase i am going to print the 3d thing

MariMakes, 2023-07-26:
Hey @HEYheyheyrey,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Have you tried reverting to the default settings?
Have you tried slicing with Support disabled? 
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.