Ultimaker/Cura #16088: Slicing failed with 3 models at the same time, but it's possible to slice them individually

menezesbruno, 2023-07-09:
### Cura Version

5.4

### Operating System

Windows 11

### Printer

Two Trees Bluer

### Name abnormal settings

Super quality 0.12mm default settings, 0.4 nozzle. Generic PLA

### Describe model location

Yes, I have to change a lot the position os the models to be able to slice them

### Describe your model

Yes, it's possible to slice them individually

### Add your .zip here ⬇️

[TWOTB_Wood_4._T2_Dakanzla_2.0.zip](https://github.com/Ultimaker/Cura/files/11996534/TWOTB_Wood_4._T2_Dakanzla_2.0.zip)
[models.zip](https://github.com/Ultimaker/Cura/files/11996540/models.zip)

GregValiant, 2023-07-11:
Thanks for the report and the files.
I can duplicate this with your model file.
Moving and rotating are two ways to fool this bug.  The third is to change the Initial Layer Height by just a bit.  at .17 Initial Layer Height it sliced.
[image]

This is a duplicate of several other reports so I'll mark it.  I'm going to leave it open as the Cura team may need more examples of the behavior.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic