Ultimaker/Cura #16094: slice failed after resizing

krypticvoyce, 2023-07-10:
### Cura Version

5.3.1

### Operating System

Windows 11

### Printer

anycubic Kobra plus 

### Name abnormal settings

15%infill, meshtool fix norms, 

### Describe model location

center

### Describe your model

scout woggle -turks head 

### Add your .zip here ⬇️

[CFFFP_scout_woggle 55.zip](https://github.com/Ultimaker/Cura/files/12001156/CFFFP_scout_woggle.55.zip)

krypticvoyce, 2023-07-10:
sliced at 60% and 50% but not at 55%

GregValiant, 2023-07-12:
Thanks for the report and the file.
It wouldn't slice for me at 55% either.  I changed the Initial Layer Height to 0.23 and it sliced.
I'll mark this as a duplicate report and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic