Ultimaker/Cura #16095: END_OBJECT error

DonStacks, 2023-07-10:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 3 V2 Neo

### Reproduction steps

Tried to print a file with the Sonic Pad, latest version.

### Actual results

[image]

### Expected results

should have printed, but stopped at after the skirt almost finished.

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/12001679/cura.log)

GregValiant, 2023-07-10:
Thanks for the report.
I'm not sure how this is a Cura problem.  There is no "iron_throne_benchy.stl" listed in the log file.
If you could elaborate on what you feel Cura did wrong it might be helpful.

DonStacks, 2023-07-10:
I’m sorry, I need to bring up the file I printed as well as the benchy.  I got the same with the calibration cube file and a couple of others.  I will try to get it to you as soon as I can.
Thanks for the response.  I really appreciate it.
Best,
Don

Sent from the all new AOL app for iOS

MariMakes, 2023-07-26:
Hey @DonStacks,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

@GregValiant is correct, we are missing quite some information here. 
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-08-10 by github-actions[bot]