Ultimaker/Cura #16098: Smart Brim checkbox reversed if using plugins that generate custom supports

Optimaximal, 2023-07-10:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Tina 2

### Reproduction steps

When slicing a model that has Custom Supports (i.e. Anti-Warping Tabs), then the Smart Brim checkbox behaviour is reversed i.e. if the Smart Brim checkbox is enabled, no Smart Brim is added.

### Actual results

When Brim is generated, old style Brim is used if Smart Brim is enabled and vice versa.

### Expected results

New Smart Brim behaviour if Box is ticked.

### Add your .zip and screenshots here ⬇️

[sberror1.zip](https://github.com/Ultimaker/Cura/files/12006009/sberror1.zip)

MariMakes, 2023-07-26:
Hey @Optimaximal,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

What a super interesting issue 🤔 
I can reproduce your issue but only with your curaproject. If I try a different model  or move the tab to a different place it seems to work as intended. 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it? 
Or do you have other examples that show the behavior that you are encountering?

Optimaximal, 2023-07-27:
I have seen the behaviour with other makes, both with and without the corner tabs.

I'm not sure if it's a side effect of my settings (i.e. maybe certain brim sizes are causing a calculation issue) but I will try to find an hour at the weekend to generate some more samples.

Can you indicate what information or config file(s) you would need to recreate my profile?

MariMakes, 2023-07-28:
Hey @Optimaximal,

Sharing your project files like you did before should be enough. 👍 
Happy hunting 💪

MariMakes, 2023-08-08:
Hey @Optimaximal,

Did you have a chance to find another file that was showing this behavior?
If it's not the case I prefer to close the ticket until other reports come in.

closed 2023-08-22 by github-actions[bot]