Ultimaker/Cura #16100: Cura won't use predefined Print Temp from Material Template

Tekklab, 2023-07-10:
### Cura Version

5.4.0

### Operating System

MacOS Ventura

### Printer

Sovol SV06 Plus

### Reproduction steps

Did a complete fresh installation on my Mac and found out, that the predefined Material Temperatures are not used within Cura. Cura always uses the 195°C no matter which material I selected.
Used the new SV06 Plus profile, but tested it with the old profile, same behavior.
Didnt had any issued in previous Cura versions on my Mac.

I sliced a model just to see, if the print temp in background are correct, only the view in Cura wrong, but that wasn't the case, the 195° are finale in the G-Code.

Looks like there is an overriding problem of the "basic" 195° print temp, because the bed temp is changing when I select a different material, only print temp will stay at 195°.

[image]
[image]

### Actual results

Screenshots from UI attached, Print Temps from different material is not imported at all

### Expected results

Print Temps changing with selecting a Material

### Add your .zip and screenshots here ⬇️

[Printtemps.3mf.zip](https://github.com/Ultimaker/Cura/files/12008562/Printtemps.3mf.zip)

GregValiant, 2023-07-11:
Thanks for the report and the file.
I cannot get it to work with your Sovol SV06 Plus printer nor with the regular SV06 so I can confirm this problem.
It works fine on my Ender and with the several other printers I have installed.  That includes the SV01 which also updates the temperature as it should.  The only one I saw that is broken are the SV06's.

I went through your Definition file and the Quality files and I don't see anything amiss.  Switching to your "CuttersPLA-SV06" custom profile - everything works as it should when changing material.
It seems that when using any of the built in quality profiles the Print Temperature just sits at 195°.

This appears to be a problem with the Sovol SV06 and SV06Plus files and I'll remove the bug label.  I'll leave this open for a while as someone else may have a better idea regarding this and also so you can respond here.  You may want to contact Sovol about this problem.

A workaround would be to create another custom Setting Profile as the one you do have (CuttersPLA-SV06) appears to work correctly when changing material.

Asterchades, 2023-07-11:
This is a fault in the Sovol printer definitions - specifically [this parent profile](https://github.com/Ultimaker/Cura/blob/main/resources/definitions/sovol_base_planetary.def.json). Note how line 16 hard-sets the print temperature (material_print_temperature) to 195, rather than letting it inherit the material properties value.

It would appear this change was merged on April 10 during the 5.4.0 beta phase via commit 044651f.

GregValiant, 2023-07-11:
Nice eye.
I wonder if should have been ""material_print_temperature": { "**default_** value": 195 } ??

Tekklab, 2023-07-11:
Thank you very much for fast response and solving. Is it possible for me to fix it for myself or should I wait for the next 4.1 ?

> This is a fault in the Sovol printer definitions - specifically [this parent profile](https://github.com/Ultimaker/Cura/blob/main/resources/definitions/sovol_base_planetary.def.json). Note how line [...]

Thank you very much for fast response and solving. Is it possible for me to fix it for myself or should I wait for the next 4.1 ?

GregValiant, 2023-07-11:
This is for Wndows.  You will need to adapt it for your MAC.
Navigate to "C:\Program Files\UltiMaker Cura 5.4.0\share\cura\resources\definitions"
Open the "sovol_base_planetary.def.json" file in a text editor.
Go to line 16 which currently is:
"material_print_temperature": { "**value**": 195 }
change it to:
"material_print_temperature": { "**default_value**": 195 }
Don't change any punctuation or indents or the file won't load.
Save the file and re-start Cura.

mstrent, 2023-10-01:
We are also experiencing this issue with a Sovol SV06. Looks like #16107 should fix. I've tested editing the file locally and it resolves.

PR #16107 (2023-11-29): Update sovol_base_planetary.def.json
# Description

Removed retraction_speed and material_print_temperature hard values. These should be defined by the material properties, not the printer definition.

Fixes #16100.

## Type of change

- [x] Printer definition file(s)

# How Has This Been Tested?

- [ ] Test A
- [ ] Test B

Untested but should be unnecessary. Basic config change.

**Test Configuration**:
* Operating System:

# Checklist:

- [x] My code follows the style guidelines of this project as described in [UltiMaker Meta](https://github.com/Ultimaker/Meta) and [Cura QML best practices]( [... cut]
closed 2023-11-29 by nallath