Ultimaker/Cura #16106: Overlapping model fails with slicing error

richievos, 2023-07-11:
### Cura Version

5.4.0

### Operating System

Mac OS Ventura

### Printer

AnyCubic Kobra

### Name abnormal settings

overlapping models

### Describe model location

haven't tried that

### Describe your model

Overlapping models

### Add your .zip here ⬇️

[AK_Skadis Screwdriver.3mf.zip](https://github.com/Ultimaker/Cura/files/12010953/AK_Skadis.Screwdriver.3mf.zip)

This is an accidental setup, but reporting because it consistently crashes. The issue is the center of the project has two overlapping models in it. It should most likely fail, but it fails with "Slicing failed with an unexpected error. Please consider reporting a bug...". Ideally it'd say "the models are overlapping, that won't work. At a minimum it should likely not suggest reporting an error.

I didn't realize I had overlapping models. It was only filing this bug that led me to debug further versus just start over.

GregValiant, 2023-07-13:
Thanks for the report and the file.
It shouldn't matter that they overlap (sometimes it's what you want).  It should still slice.
I moved the other models out and just left the two overlapping models and it still wouldn't slice.  When I changed the Initial Layer Height to 0.23 if did slice.
I agree that this is a bug and it appears to be one that has been reported previously.  The "overlapping models" I think is just a red herring.
The Cura team will take a look.

BTW...The tabs you are using appear to be there to keep the edges from warping.  In the project file the tabs are set to "Print as Support".  If you are using them to keep warpage down then I suggest you go to Per Model settings and change each tab to "Print as Normal Model".

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic