Ultimaker/Cura #16108: Retraction Settings not applied for Multiple Extruders

Jasey1410, 2023-07-11:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Custom (Marlin)

### Reproduction steps

1. Import a model with multiple parts (one main part and 2 internal parts used as modifiers) all in one *.3mf file, so that the relative orientation is correct
2. Set Extruder 2 as the Extruder to print all three parts, but change INNER_WALL_EXTRUDER to Extruder 3.
3. Change Retraction Distance and Speed for each Extruder (Extruder 2: Retraction Distance=0.4mm, Extruder 3: Retraction Distance=10mm)
4. Slice

I attached two *.3mf files and the gcode-file, in case you want to try to reproduce the bug. First one is just the model, second one is the project file.
Cheers,
Jan
[Cura Bug_Retraction.zip](https://github.com/Ultimaker/Cura/files/12013257/Cura.Bug_Retraction.zip)

### Actual results

Look at GCode and see, that even if Extruder 3 is used for printing the inner walls, the retraction settings (E-0.4 instead of E-10) are used, which are the Extruder 2 settings !! It seems like if I select Extruder 3 to print one of the internal parts completely (not only the inner walls), the retraction settings for Extruder 3 are used. If I select only a certain feature of a part to be printed with Extruder 3, Cura still takes all the Extruder specific settings for the Extruder which is selected as the "main extruder" (in my case: Extruder 2) to be used for printing that part. 
I only checked for the Retraction settings, so not sure if other Extruder specific settings are also not changed.
I have the feeling that the Retraction settings are only checked per printer part, not per feature (Inner_Wall, Outer_Wall, Infill, etc.).

### Expected results

I expect that everytime a different Extruder is used for printing, even if it is just a small feature like inner walls, outer wall, infill, support, etc., the extruder specific settings are used and not the overall selected Extruder's settings are applied to all printing_features of that part.

I hope that it is possible to change this, because I highly appreciate the possibility to print certain detailed features with different Extruders :) It is very powerful and not possible with other Slicers ;)
Cheers,
Jan

### Add your .zip and screenshots here ⬇️

[Cura Bug_Retraction.zip](https://github.com/Ultimaker/Cura/files/12013300/Cura.Bug_Retraction.zip)

GregValiant, 2023-07-12:
Thanks for the report and the files.
Just so we are on the same page - Extruder numbering in Cura is base1 and extruder numbering in gcode is base0.

What I see in the Gcode is that T2 (Extruder3) is being used for the Wall-Inner features.  That is as it should be.  But the retraction distance does appear to be wrong as T2 is using the T0 retraction distance of 3.53 rather than the 10mm it is set to.  The extruder stack confirms that T2 "retraction_amt" is set to 10mm.  So it appears that the variable is being stored correctly but implemented wrong.
Opening the project in 4.13.1 the retractions are correct.

The Cura team will take a look.

Jasey1410, 2023-07-12:
Hi, thanks for having a look into it.
Yes, I am aware that in Cura Extruders are base1 and in gcode base0. When I talk about Extruder 3, it is T2 in the gcode 👍 
In previous versions of Cura (also in 4.13.1, as can be seen in the image I attached), Cura would not adjust the Line-Width according to the Extruder's nozzle diameter, if a certain feature (like Wall-Inner in this case) is selected to be printed with a different extruder than the rest of the part. This was fixed with 5.4.0.
For some reason the Skirt is printed by both Extruders used during the print, even if only one is specified to be used for the Skirt. This is actually also still happening in Cura 5.4.0 I just noticed, but does not bother me too much. The Line-Width for the Skirt printed by Extruder 3 in Cura 4.13.1 is correct for some reason, whereas for Inner-Wall, it's not.
Cheers, 
Jan

[image]

GregValiant, 2023-07-12:
Extruders that are enabled and used in a print seem to always have at least a line in the skirt.  It gets the flow going as that is the point of the skirt.  At least that is what it appears to be.

When I set the Outer Wall Line Width to 0.8 the Outer Walls do go down at that line width.
I have a  macro for Excel that will do an analysis of gcode files.
I sliced the model in 5.4.0...
This is from the skirt.  I've highlighted the line width that the Volume of the E will produce at a 0.20 layer height.
935	0.03006		0.03006	;LAYER:0	15	0.07230	0.90389	0.06026	1.19987	**0.40**	SKIRT
This is from the Outer Wall which I had set to 0.80
942	0.85135		0.85135	;LAYER:0	50	2.04774	12.79837	0.25597	7.99999	**0.80**	WALL-OUTER
Because of Cura's variable line width, I have to extrapolate the line width from the known layer height, the volume of the extrusion, and the length of the extrusion.  So in this example everything seems to have worked as it should.
Switching the inner-wall line width to 0.80 still provides a good result.

So what I see of this bug is that regardless of the extruder that is working, the Retraction Amount is the Retraction Amount of the extruder that the model is "set to" as opposed to being the retraction amount that was assigned to that extruder.  If the model is set to Extruder 1 then all extruders will use the E1 retraction amount.  If it is set to Extruder 4 then all extruders will use the Extruder 4 Retraction Amount.
I am seeing the same behavior in 4.13.1 so it would appear that this has been a problem for a while.

Jasey1410, 2023-07-24:
Will this be implemented in the next Release or is there an option to get a pre-release with this bug being fixed ?
I'm just asking, because I'm working on a post-script right now and it would be nice to know if I should wait for a version with this bug being fixed or trying to work around it somehow.
Just an approximation: 1 week, 1 month or more than that would be helpful.
Cheers