Ultimaker/Cura #16112: slicing failed

Foggy3DPrints, 2023-07-11:
### Cura Version

5.4

### Operating System

windows 10 pro

### Printer

anycubic vyper

### Describe model location

no

### Describe your model

no

### Add your .zip here ⬇️

[Skulldroid_Skull.zip](https://github.com/Ultimaker/Cura/files/12018074/Skulldroid_Skull.zip)

GregValiant, 2023-07-13:
Thanks for the report and the file.
It wouldn't slice until I rotated the model 90° in Cura.
[image]

I'll mark this as a duplicate and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic