Ultimaker/Cura #16113: Slicing fail

racheljmforster, 2023-07-11:
### Cura Version

5.3.1

### Operating System

MacOS Monterey

### Printer

Creality Ender 3 V2

### Name abnormal settings

Brim, tree supports. Yes.

### Describe model location

Its a crescent shape with the center to the front left. It does slice if I rotate it.

### Describe your model

Yes, i've sliced it before. Its a flat surface so not water tight. No I haven't tried Meshtools

### Add your .zip here ⬇️

[facecut.3mf.zip](https://github.com/Ultimaker/Cura/files/12021227/facecut.3mf.zip)

GregValiant, 2023-07-13:
Thanks for the report and the file.
This sliced for me with no problem.  It might be a Cura for MAC issue.
I'll mark this as a duplicate and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic