Ultimaker/Cura #16114: CURA 5.4.0 Slice Problem Anet A8+

auroraits, 2023-07-11:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Anet A8+ 

### Name abnormal settings

Marlin / 0.2mm Nozzle

### Describe model location

Anywhere fails

### Describe your model

Worked fine few times

### Add your .zip here ⬇️

LINK https://auroraits.com/AA8P_Master_Chief_CANTSLICE.zip

GregValiant, 2023-07-13:
Thanks for the report.
I didn't find any errors in the model.  I can't get this to slice in 5.4.0.
It sliced in Cura 4.13.1, PrusaSlicer and IdeaMaker.
I'll mark this as a duplicate and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic