Ultimaker/Cura #16115: Initial Top Layer Layers

wuxn3d, 2023-07-11:
### Is your feature request related to a problem?

Speed, acceleration, and jerk settings should be adjustable for the first top layer. As top layers are printed over infill, sagging and curling may occur, leading to nozzle dragging.

### Describe the solution you'd like

Add the following settings:
Initial Top Layer Speed
Initial Top Layer Acceleration
Initial Top Layer Jerk

### Describe alternatives you've considered

Extra Skin Wall Count adds significant slicing time and doesn't address the issue entirely. Increasing infill percentage helps but adds more print time and required material.

### Affected users and/or printers

Everyone, especially for users trying to print fast.

PhilBaz, 2023-07-12:
You can use bridge settings to control some of this. By changing the infill support threshold to something high you can control 'internal' bridging. Check it out.

wuxn3d, 2023-07-13:
Thank you Phil. If you ever need a job, we're located in Colorado USA.

PhilBaz, 2023-07-13:
@wuxn3d 

Thank you. If I was in the area id come have a chat :)

Good luck.