Ultimaker/Cura #16118: Error slicing

commanderferro, 2023-07-12:
### Cura Version

5.4.0

### Operating System

MacOS Ventura 13.4.1 

### Printer

Creality Ender 3 V2

### Name abnormal settings

It doesn't slice

### Describe model location

Slice when i change the size, under 50%

### Describe your model

yes, with the cura version before this

### Add your .zip here ⬇️

[CE3E3V2_gizmo.3mf.zip](https://github.com/Ultimaker/Cura/files/12027729/CE3E3V2_gizmo.3mf.zip)

GregValiant, 2023-07-12:
Thanks for the report and the file.
I didn't have any trouble slicing the project so I can't duplicate the issue.  I checked the models with Cura's Mesh Tools and 4 models are "Not watertight and may not print correctly".  That might be having an effect on your system.

I'll leave the bug label on this and the Cura team will take a look.

[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0