Ultimaker/Cura #16119: Eject SD used to work in 4.x but doesn't in 5.x

rpweaverjr, 2023-07-12:
### Is your feature request related to a problem?

The old Cura used to allow Eject from the Dialog box after Save. This no longer works (EVER). I always get that the device MAY be in use by another program even when nothing else may be running. Please check into getting this back into 5.x, They Eject dialog box is useless and annoying since it doesn't work.
[image]

### Describe the solution you'd like

Make it work like it used to.
[image]

### Describe alternatives you've considered

I really don't want to give up the new features for this but having to go to File Explorer every time to Eject which always works, BTW, is very annoying.

### Affected users and/or printers

It saves a few steps and avoids physically ejecting a card when it might not be ready.

GregValiant, 2023-07-12:
I'm using Windows 10 and the feature still works for me in 5.4.0.  I don't recall it not working in other versions either.
What operating system are you using?  Cura makes calls to the OpSys when dealing with the drives.

rpweaverjr, 2023-07-12:
Windows 10. It used to work in Cura 4.x. on this laptop. I'm running 5.4. Is there a setting that would fix this?

Just checked 4.13.1 and it didn't work there either. I don't have anything older installed. I actually don't remember when it quit working but it hasn't worked since.  I've been annoyed by this for some time.  Maybe it is something Microsoft did.

Like I said,  File Explorer doesn't have an issue ejecting the drive but Cura always does. Looks like it is still a mystery.

Thanks,
Richard Weaver

Sent from my Verizon, Samsung Galaxy smartphone

GregValiant, 2023-07-12:
I'll change this to a bug report.  Maybe one of the developers has run into something like this and will know how to fix it.