Ultimaker/Cura #16121: Custom Support Cylinder ("F") Causes slice failure no matter what I try.

justinb1033, 2023-07-12:
### Cura Version

5.4.0-beta.1

### Operating System

win 10

### Printer

Crealty CR-10

### Name abnormal settings

Custom Support Cylinder ("F")

### Describe model location

no

### Describe your model

Model will slice without custom supports. Even adding 1 custom support causes error.

### Add your .zip here ⬇️

[CCR10_Hex3D_PredatorViking_P4.zip](https://github.com/Ultimaker/Cura/files/12028968/CCR10_Hex3D_PredatorViking_P4.zip)

GregValiant, 2023-07-12:
Thanks for the support and the file.
I didn't have any trouble slicing your project.  Try moving or rotating the models.
[image]

This type of bug has been reported previously.  I'll mark this as a duplicate.

BTW the maximum you can print at 0.2 layer height and 0.6 line width is 71° overhang angle.  You have it set to 80° and that would allow some extrusions to sag.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic