Ultimaker/Cura #16123: Slicing into layers is not performed

Kotemorte, 2023-07-12:
### Cura Version

5.4.0

### Operating System

Win 10

### Printer

ender 3 s1 plus

### Name abnormal settings

Settings absolutely normal stock settings from eSUN pla+ gray

### Describe model location

centered

### Describe your model

head droid

### Add your .zip here ⬇️

[CE3S1P_Head.zip](https://github.com/Ultimaker/Cura/files/12031412/CE3S1P_Head.zip)

Kotemorte, 2023-07-12:
by the selection method, I realized what the problem is, he can't make a cut with tree-like supports for some reason... previously, he could do this, I repeat the stock settings from eSUN pla+ gray the most interesting thing is that the usual support from slices and tree-like refuses

GregValiant, 2023-07-12:
Thanks for the report and the file.
I didn't have any trouble slicing your project so I'm unable to duplicate the problem.
[image]

Try moving the model or maybe rotating the model.  Sometimes changing the Initial Layer Height a little bit (like from .20 to .23) can fool this bug.
This behavior has been report previously so I'll mark this as a duplicate and the Cura team will take a look.

closed 2024-04-03 by HellAholic