Ultimaker/Cura #16125: Error Slicing

chrisbensen, 2023-07-12:
### Cura Version

5.3.0

### Operating System

MacOS 13.4.1 (22F82)

### Printer

S5

### Reproduction steps

1. slice the attached model
2. use the settings in the screen shot attached
3. get the error also attached in the screen shot, it fails to slice.

### Actual results

get "slicing failed" error

### Expected results

I expected it to slice

### Add your .zip and screenshots here ⬇️

[Archive.zip](https://github.com/Ultimaker/Cura/files/12033088/Archive.zip)

chrisbensen, 2023-07-13:
The problem is tree supports.

MariMakes, 2023-07-26:
Hey @chrisbensen,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

The new Cura 5.4 contains the fix for this for a number of slicing crashes, especially those related to tree supports. 

Could you test if that's the case for you as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

If you still run into a crash, please double-check if you shared the correct file and let us know.
I'll take another look at what is going wrong. 👍

closed 2023-08-10 by github-actions[bot]