Ultimaker/Cura #16130: Slicing fehlgeschlagen

Yoda1963, 2023-07-13:
### Cura Version

5.4.0

### Operating System

MacOs Ventura

### Printer

Anycubic Kobra

### Describe your model

nein

### Add your .zip here ⬇️

[AK_BabyGrootPencilHolder.3mf.zip](https://github.com/Ultimaker/Cura/files/12039841/AK_BabyGrootPencilHolder.3mf.zip)

Yoda1963, 2023-07-13:
Wenn ich das Modell in original Größe belasse, kann ich es slicen. Verkleiner ich das Modell bekomme ich die Fehlermeldung: Slicing fehlgeschlagen.

GregValiant, 2023-07-13:
Thanks for the report.
Try changing the Initial Layer Height by a little bit.  If you are using 0.20 then try 0.23 or 0.19.  That will sometimes fool the bug.
I'll mark this as a duplicate.

Yoda1963, 2023-07-13:
Vielen Dank, es hat funktioniert.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic