Ultimaker/Cura #16131: wrong temp for selected extruder

firepower9966, 2023-07-13:
### Cura Version

5.4.0

### Operating System

Linux Mint 21.1

### Printer

Custom E3D Volcano (Geetech A10M originally)

### Reproduction steps

I have 3 extruders in firmware and use the 3rd one as direct drive. 
extruder 1 and 2 are not used and disabled
it is a single nozzle heater.
5.4 cura is using hotend print temp of extruder 1 for the printing temp of extruder 3 

### Actual results

to get correct print temp for extruder 3 i have to set it for extruder 1 

### Expected results

expected result is setting extruder 3 temp and in slicer and extruder prints with that temp

### Add your .zip and screenshots here ⬇️

[Cura 5.4 temp error.tar.gz](https://github.com/Ultimaker/Cura/files/12039955/Cura.5.4.temp.error.tar.gz)

GregValiant, 2023-07-13:
Thanks for the report.
Without a project file we can't tell what is going on.  Have you disabled "Extruders Share Heater" and "Extruders Share Nozzle"?

firepower9966, 2023-07-13:
> —conf includes

I attached screen shot and project file  cura_5.4_temp_error.zip
I have installed my custom printer files on all previous versions of cura
and had no issues
I even done a clean install of cura with my custom files which i have also
attaches resource.zip
I have not disabled "Extruders Share Heater" and "Extruders Share Nozzle"
and i sure my config includes "shared heater single nozzle" as it used to
be a tick box option on previous versions of cura
printer machine setting.

thanks for your prompt reply, hope the extra info helps you solve this
issue.
thanks Neil

firepower9966, 2023-07-13:
> A project file is a 3mf file. I'm not familiar with a ".error.tar.gz" file.
> Load your model and set Cura up as you normally do. Use the "File | Save
> Project" command to create the 3mf file. Zip it and post the zip folder
> here.
> There has been a bug report regarding the retraction distances being used
> by multi-extruder machines. This may or may not be related.

tar.gz is the default zip file for linux, you should be able to unzip it
also, but I also resent the files as zip in another email.

With my cura settings you see extruder 1 is set to 210C , extruded 2 is set
to 220C and Extruder 3 is set to 240C which is the extruder I am  printing
Petg with.
The G code shows T2 is printing with 210C when it should be 240C

GregValiant, 2023-07-13:
I installed a Geeetech A20T.  I noticed that in the StartUp Gcode there are temperature lines M104 S215 and M109 S215.
I removed those lines and allowed Cura to set the temperatures.
I was able to get the project file but I only got your model and not your printer so I have to guess at some of this.
With those temperature lines removed I don't see a problem in the gcode.  T2 is the only extruder used in the print and it prints at 240 (my setting).  That was with T0 and T1 disabled.
So far I'm not able to duplicate the problem.

firepower9966, 2023-07-13:
[cura 5.4 temp error.zip](https://github.com/Ultimaker/Cura/files/12042468/cura.5.4.temp.error.zip)

firepower9966, 2023-07-13:
here are my files , i attached via email but looks like they not attach so upload here in github

firepower9966, 2023-07-13:
[resources.zip](https://github.com/Ultimaker/Cura/files/12042500/resources.zip)

firepower9966, 2023-07-13:
here is my custom printer  files

closed 2023-07-13 by firepower9966

GregValiant, 2023-07-13:
Put a semicolon in front of the M104 and M109 temperature lines of the E3D printer Startup Gcode.  Let Cura handle the temperatures.   Removing the lines completely is also an option.

The extruders and temperatures are defined correctly (as witnessed by the M104 S240 lines later in the gcode) but the Replacement Patterns in the StartUp Gcode are simply replaced.  They aren't "read".  What might be happening is that the Extruder Stack isn't being consulted for which extruder is active and Cura is simply entering in the temperature of the bottom extruder which is T0. 

I'm going to remove the bug label from this and close it.  You have a 3rd party printer with a 4th party hot end that was installed by a 5th party so this isn't really a problem with Cura.  I hope you understand.  The fix above should work.