Ultimaker/Cura #16133: Wrong commands in gcode for Anycubic Kobra Max

syrandhar, 2023-07-13:
### Cura Version

5.3.1

### Operating System

Windows 10

### Printer

Anycubic Kobra Max

### Reproduction steps

added the printer to ultimaker cura
made a model
options say only "lower bed" for printhead movements... there is no "raise printhead" because the Kobra Max cant lower bed

### Actual results

the model has lines in the upper surface from printhead movements

### Expected results

the printhead should be raised for movement because the lower bed movement is not available on the printer

### Add your .zip and screenshots here ⬇️

G1 Z15.0 F1200 ;Move the platform down 15mm

thats the first line i find in the gcode file mentioning it

GregValiant, 2023-07-13:
Thanks for the report.
Your printer definition file comes from either AnyCubic or a member of the community.  It was submitted to UltiMaker for inclusion in Cura.  UltiMaker has no way to check on another manufacturers printers.

That being said, what you see there is a generic term that almost all printers use regardless of whether the bed moves up and down or the print head moves up and down.
On some printers G1 Z15.0 means move the printhead away from the bed.  On other printers it means move the bed away from the printhead.  Either way - at the end of the move the gap between the nozzle and the bed will be 15mm.

I looked at your startup gcode in the Kobra definition file and I don't see anything wrong with it.  The printer auto-homes and the nozzle moves away from the bed (or the bed moves away from the nozzle) 15mm prior to any other movement.

I'm going to remove the bug label and close this because what I see is:  expected behavior and verbiage, or, it's an AnyCubic problem (but I don't really see that).
I hope you understand.

closed 2023-07-13 by GregValiant

syrandhar, 2023-07-22:
i just looked through the whole gcode-file.. there are only one more commands with "Z".. always when the next layer starts...
shouldn't there be "Z" commands with every printhead movement?

GregValiant, 2023-07-22:
The position of any axis stays where it is until another movement command mentions the particular axis.
G1 X100
G1 Y100
G1 Z50
G1 F7200
from a strictly position viewpoint is the same as
G1 F7200 X100 Y100 Z50
Lines like retractions/primes are in the form of: G1 F2700 E23.34567 as are Z-hops G1 F2700 Z23.34
In both cases the X and Y aren't mentioned and so they stay where they are.