Ultimaker/Cura #16134: articulated Umbreon

Betolfo, 2023-07-13:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Anycubic Cobra Plus

### Name abnormal settings

Normal settings

### Describe model location

still at the center off the buildplate

### Describe your model

is the first time that I try to print

### Add your .zip here ⬇️

[Umbreon-Bug.zip](https://github.com/Ultimaker/Cura/files/12041563/Umbreon-Bug.zip)

GregValiant, 2023-07-13:
Thanks for the report and the file.
The model contains 1,057,962 triangles.  That is incredible for a model that small and there is no way that the FDM process can print anything that close.  I'm not sure if the triangle density is the entire problem, but when I simplified the model it sliced.
This is the simplified model (39,964 triangles)

Your full density model sliced when I set the Initial Layer Height to 0.23.  That is a known bug so I'll mark this as a duplicate.
[image]

Betolfo, 2023-07-14:
Thanks for your help, well I'll ready bought the model at my minifactory 3d.  I'm no so sure how to simplify the design.
I'm glad there in nothing wrong whit the Cura slicer. Thanks again.

[image]<https://user-images.githubusercontent.com/64202104/253405268-24e7766a-2cd7-4af5-99d6-e6d55790eb7f.png>

GregValiant, 2023-07-14:
You're on Windows so you can go to the Microsoft App Store and download MS 3D Builder.  It has a lot of utilities for altering and repairing model files.  One of them is a "Simplify" tool.  The model went from over 1 million triangles to a much easier to deal with 39k triangles.  I used 3D Builder to do that.

Betolfo, 2023-07-14:
Thank you very much. I'll use it.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0