Ultimaker/Cura #16135: City not slicing - unexpected error

amolino2, 2023-07-13:
### Cura Version

5.3.1

### Operating System

Windows 11

### Printer

Elegoo Neptune 3 Max

### Name abnormal settings

Does not print on default profiles, changed a number of things but still within normal settings

### Describe model location

No

### Describe your model

I have not sliced it successfully before. I have not

### Add your .zip here ⬇️

[EN3MAX_Solid_Luthadel.zip](https://github.com/Ultimaker/Cura/files/12044582/EN3MAX_Solid_Luthadel.zip)

GregValiant, 2023-07-14:
Thanks for the report and the file.
There are errors in the model.  That can effect slicing.  I was unable to slice your project in either Cura or PrusaSlicer.
I'll leave the bug label on this for now but that model is definitely part of the problem.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic