Ultimaker/Cura #16139: Enable Support Brim causes problems if "None" is selected for Build Plate Adhesion

ZedOud, 2023-07-14:
### Cura Version

5.4.0

### Operating System

MacOS Ventura 13.4.1

### Printer

AnkerMake M5, Flsun V400

### Reproduction steps

Reproduction steps:
1. With either Tree or Normal support, select Enable Support Brim.
2. Select "None" for Build Plate Adhesion.
3. Now, the first layer of the support will not be generated (the 2nd layer will print in mid-air).

Alternative step 2:
2a. Select "Skirt" for Build Plate Adhesion.
2b. Set Skirt Line Count to 0
2c. Set Skirt/Brim Minimum Length to 0

### Actual results

It should not remove the entirety of the first layer of support.

Sometimes, it does generate a quirky miniature support that has around 25% the footprint of the actual support on Normal support. This may depend on the Support Brim Width value. It consistently occurs if no part of the model is touching the build plate. (Example A project file.)
If no part of the model is touching the build plate, then with Tree support you will get a normal first layer without a support brim. 
If at least some part of the model is touching the build plate, then this bug should occur. (Example B project file.)

### Expected results

This pattern should either refuse to generate a Support Brim, or ignore the Build Plate Adhesion settings category and print a Support Brim.

### Add your .zip and screenshots here ⬇️

[Examples.zip](https://github.com/Ultimaker/Cura/files/12046279/Examples.zip)
Example A - Full Preview:
[image]
Example A - 1st Layer Preview:
[image]

Example B - Full Preview:
[image]
Example B - 1st Layer Preview:
[image]

Asterchades, 2023-07-14:
This is a known issue, per the release notes for 5.4.0.

> Support Brim is missing if the Buildplate Adhesion is set to None. [We are tracking reports here](https://github.com/Ultimaker/Cura/issues/14983).

Duplicate #14983.

closed 2023-07-14 by Asterchades

ZedOud, 2023-07-14:
> This is a known issue, per the release notes for 5.4.0.
> 
> > Support Brim is missing if the Buildplate Adhesion is set to None. [We are tracking reports here](https://github.com/Ultimaker/Cura/issues/14983).
> 
> Duplicate #14983.

It may be a duplicate, but as far as I can see, no one has found out the 2nd method of producing the bug:
Alternative step 2:
2a. Select "Skirt" for Build Plate Adhesion.
2b. Set Skirt Line Count to 0
2c. Set Skirt/Brim Minimum Length to 0